PetoiCamp/OpenCatEsp32-Quadruped-Robot #1: Esp32

marcotchella, 2022-03-01:
Hi 
I´m learning robotics with this fantastic project.
I´m interested in Biboard and ESP32 version
In documentation i see that imu libarie need changes and you offer a modified version
The question is how to modifie the librarie or get the version that you offer
Thank you

https://docs.petoi.com/biboard/biboard-v0#3.2.2-imu

borntoleave, 2022-03-01:
I developed a new IMU routine based on this repo https://github.com/ElectronicCats/mpu6050/, which is compatible with ESP32 and ATmega328P. 

The current repo is just a walking demo to show every peripheral can work. It is going to have a big update soon. I have just finished a complete revision to give the ESP32 board (BiBoard) a superior performance and integrity than the previous OpenCat firmware on ATmega328P (NyBoard). You may follow our site at petoi.com or my social @petoicamp for the release announcement. I may put the new version in the grand [OpenCat repo](https://github.com/PetoiCamp/OpenCat). 

I'm attaching the changelog of the new firmware so you can expect what new features to come:
# Change Log

## Mar 1, 2022
* Support IOS Ble app
* Correct the angle range of m token (use int instead of int8_t).

## Feb 26, 2022 
* Correct an error when reading commands. The remaining serial commands should be checked within the token's available condition. 

## Feb 22, 2022
* Change skillList to a pointer and init it in setup().
* Correct the lastCmd/newCmd logic of the initial posture.

## Feb 19, 2022
* Add macro for GYRO.
* Make the robot enter the joint calibration state (different from initialization) if it is upside down. Otherwise, start up normally.

## Feb 8, 2022
* Reduce the odds of random noise when shutDown servos by detaching them then re-attach. 
* Improve the sound for falling over.

## Feb 7, 2022
* Set the bootup posture to calibrate state, for convenient installation. Disable gyro and auto behavior until first user input. 
* Optimize servo angle range.
* Merge the token FOLD with regular skills.
* Tune the pee, hi, and balance skills.
* Optimize low battery checking logic. Safely transform to rest posture then shut the servos. 
* Tune LED.
* Tune adjustment damper.
* Remove the delay between postures.
* Add special sound for falling over.

## Feb 6, 2022
* Add a 'v' token to print the MPU for once and a 'V' token to print the MPU for every loop. 
* Fix a bug when reading IMU acceleration data.
* Add comments the difference between REAL_ACCELERATION and WORLD_ACCELERATION.
* Correct the servos' joint angle range.
* Add a low voltage alarm for BiBoard2.
* Make NeoPixel blink with motion for BiBoard2.

## Feb 5, 2022
* Add T_PRINT_GYRO 'V' to toggle printing of Gyro data. 

## Feb 2, 2022
* Merge PCA9685 PWM driver in the main code to be activated with proper macro.
* BiBoard2 works!

## Feb 1, 2022
* Implement writeAngle() outside the Petoi_PWMServoDriver Class.

## Jan 29, 2022
* Fix the infinite loop of "play dead - recover" and "m (or other non-skill token) - recover".
* Use hardware PWM tone library to generate beep and improved the music quality.

## Jan 27, 2022
* Create a token 'z' to turn on/off the power saver and random behaviors.
* Create a token 'r' to change the adaption directions on a ramp.
* Define fold and rotate behavior for cub.
* Add I2C detector and macro for IR.

## Jan 26, 2022
* Write the log up-to-date.
* Restructure T_SKILL reaction to simplify the control flow.
* Fix bug in power saver.
* Improve power saver and randomMind logics.
[... 65 lines cut ...]
## Jan 9, 2022
* Move dependent libraries into the project folder to help new users.
* Add calibration routine for MPU6050 and save the offsets in EEPROM during initialization. 
* Add a "birthmark" to know if a board has been initialized so that it will boot up directly next time.

## Jan 8, 2022
* Solve the detached servo problem by re-attaching the corresponding channels to continue other parts. 
* Joint calibration routine works. 
* MPU works but without calibration. 

## Jan 1, 2022
* Fix the bug when switching to behaviors by correcting the servo indexes on a 12 PWM board.

## Dec 30, 2022
* Behavior partially works.
* Add a boolean to check if IMU is updated and decide if to recalculate the balancing adjustment values.

## Dec. 28, 2022
* Servos can move.

## Dec 16. 2022
* Improve the code and demo's doc for serialMaster.

## Dec. 10, 2022
* Fix the esp32-hal-i2c-slave.c bug that locks IMU at booting stage.

marcotchella, 2022-03-01:
Hi @borntoleave. Thank you for the answer.
I´m following social and site. I will wait upgrades in ESP32 platform.
Thank you

marcotchella, 2022-03-01:
Another question
The file **esp32-hal-i2c-slave.c** is available?
Thank you

https://docs.petoi.com/biboard/biboard-configuration#esp32-hal-i2c-slave.c

borntoleave, 2022-03-02:
The header file will be included in the new release. It corrected a few
parameters in the current official package. We have also submitted the
issue and hope it could be solved in their next update.

closed 2023-03-08 by borntoleave