PetoiCamp/OpenCatEsp32-Quadruped-Robot #17: testBLE_uart sketch fails to compile

FlexoTim, 2024-06-11:
The sketch contains an stl::string variable which is illegal.

borntoleave, 2024-06-14:
What compiler are you using? Arduino IDE would generate a warning but still compiles. I've modified it to String to clear the warning.

FlexoTim, 2024-06-14:
I'm using the Arduino IDE v2..3.2. I also use the arduino-cli but not on this issue.
The compiler is: xtensa-esp32-elf-g++ (crosstool-NG esp-12.2.0_20230208) 12.2.0
The error I get should is not be compiler dependent:

std::string rxValue = pCharacteristic->getValue();

will never compile because there's no conversion from String to stl::string.

There's another warning on

char * genBleID(int suffixDigits = 2, char prefix[] = CONNECTION_NAME)

because CONNECTION_NAME is a string literal, and the pointer isn't const. that's a different issue.

este-este, 2024-06-15:
I **_am_** able to compile the testBLE_uart.ino using either Arduino IDE 2.2.1 or Arduino IDE 1.8.19.  Target Board (under Tools menu) is the Arduino ESP32 > ESP32 Dev Module.  I get no warnings.

In Windows 10, this is the path to the tool which indicates what compiler version I am using:  ...\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0

FlexoTim, 2024-06-15:
I'm attaching the error log for completeness.
[Issue 17 log.txt](https://github.com/user-attachments/files/15845199/Issue.17.log.txt)

I'm at a loss for words here. This needs to be nailed down. From my perspective, this is a C++ language issue. I would argue a compiler that accepts the code in question is broken.

The code in question (which is the code that git says was changed) is
      std::string rxValue = pCharacteristic->getValue();

The function BLECharacteristic::getValue returns a String object.
The String class provides no conversion operator to a type that can be used to initialize a std::string. This is the error.

The statement can be simplified to make it more direct:
    std::string rxValue = String("abcd");

If the compiler doesn't declare this as an error then I don't know what it's doing.

borntoleave, 2024-06-15:
Have you been able to try my fix, which was uploaded yesterday?
https://github.com/PetoiCamp/OpenCatEsp32/commit/5be1d6b50e6b225b4c0da11a58b2bb7439f522d2#diff-ff52102038a6d9e40a9227b172b6dd4a49b2ed6ffb50b20a45e79e2711b2d3d3

FlexoTim, 2024-06-15:
Yes. Sorry for the delay; distractions on my end
 I got the changes from the repository and successfully built

Sketch uses 1115061 bytes (85%) of program storage space. Maximum is 1310720 bytes.
Global variables use 35004 bytes (10%) of dynamic memory, leaving 292676 bytes for local variables. Maximum is 327680 bytes.

The issue is resolved from my POV.
Thanks

borntoleave, 2024-07-29:
I found a bug in sending long 'K'-token skills via the mobile app after the modification. 
Though 
**String rxValue = pCharacteristic->getValue().c_str();**
and 
**std::string rxValue = pCharacteristic->getValue();**
both compile and should be the same, there's some unknown error in the conversion results the machine's main program break.

FlexoTim, 2024-07-29:
I don't understand.
testBLE_uart doesn't send skill data.

I don't see how

std::string rxValue = pCharacteristic->getValue();

can ever legitimately compile. There's no conversion from String to std::string.

borntoleave, 2024-07-31:
It's used in the main program to receive commands from the mobile app. The
bug raises when receiving long skill data after token 'K'.

I haven't figured out why the String method fails. So, I rolled back to the
std::string method.

FlexoTim, 2024-07-31:
I'm doing this privately so we can get on the same page in terms of
communication.

I would call that a related issue but not an issue in testBLE_uart. This is
your repository and process, so it's up to you but I would log the error as
a related issue against OpenCatEsp32. I just want the error to be as
specific as possible.

Is this happening in the existing code or new code? In other words, can I
replicate this behavior with my code base? I need more detail  about how to
replicate the issue or analyze the code.

I'm going  to make an assumption that this is related to the new arm
attachment. The new DOF and weight affects  skills and gait, etc. and may
uncover new or old issues. Do I need to look at a different  code branch?

I think we want to try to isolate the issue to either the BLE interface or
the skill data processing.

I say this because in order for me to do something here  I need to see the
new code and see if I can spot the error, or somehow replicate the behavior.

I understand reverting the code back to std:string to see if that's the
cause. What was the result? Same or different behavior? I would suspect the
same.

In any case, I'll help as best I can but I need something to work with,
i.e., some code.

We can communicate via email rather than through the repository or continue
through the repository. It's up to you.

Tim

borntoleave, 2024-08-01:
Hi,
I have isolated the problem to that specific line of code:
[image: Screenshot 2024-08-01 at 14.48.42.png]
The main program crashes when sensing 'K' token commands with large numbers
of skill data. Switching between these two lines reproduces and resolves
the problem.

Some bytes must be changed when converting between std:string and Arduino
String.

FlexoTim, 2024-08-02:
I can't see the screenshot file.

I can't replicate the issue without the data used to produce it. What's necessary to replicate what "long skill" command means?

I'm suspicious about how the write of the BLE characteristic is involved with reading skill data. Does this happen with the UART? If it's a problem with the program memory footprint then it will resurface in some other location. If it's a conversion issue, then it should be testable outside of the BLE API.

borntoleave, 2024-08-03:
I'm attaching the image here:
[image]

A test command can be [imported](https://docs.petoi.com/mobile-app/controller#import-new-skills-as-a-customized-button) as a customized mobile app button:
[Bittle_MoonWalk.md](https://github.com/user-attachments/files/16477928/Bittle_MoonWalk.md)