PetoiCamp/OpenCatEsp32-Quadruped-Robot #18: use raw voice input

ysj1173886760, 2024-06-23:
hello, do we have options not parse the voice collected by microphone into command, but output it directly?

borntoleave, 2024-06-24:
The microphone only returns the mapped strings listed at
https://docs.petoi.com/extensible-modules/voice-command-module#id-2.-use-the-predefined-voice-commands

ysj1173886760, 2024-06-24:
get, thanks for your reply. Guess i need another microphone to collect additional sounds

closed 2024-06-24 by ysj1173886760

este-este, 2024-06-24:
I have a related question.  Can those mapped strings be changed programmatically (perhaps by reflashing the memory in the voice module)?  Basically, I 'd like to programmatically trigger voice feedback (e.g. have the voice module say "Object detected", triggered by a programmatic detection of an object on one of the sensors.  Would be pretty cool to be able to do this...)