PetoiCamp/OpenCatEsp32-Quadruped-Robot #22: Check esp32 library version

FlexoTim, 2024-08-02:
A plea.

I've been tripped up a number of times by the esp32 library version issue. OpenCatEsp32 requires the esp32 library version of 2.0.12, which is obsolete and no longer supported by Espressif. The Arduino IDE is constantly asking to update to the current version (3.0.4 now) and if you do then you have trouble with the OpenCatEsp32 project. It would helpful if OpenCatEsp32.h would check the library version, e.g.,

`#if ESP_ARDUINO_VERSION != ESP_ARDUINO_VERSION_VAL(2, 0, 11)
#error esp32 v 2.0.12 required
#endif
`
```
`

este-este, 2024-08-02:
Personally, I defer to the Petoi team as to the return on investment to be had by updating the OpenCatEsp32 code to use the current esp32 library.  I am not sure what support Petoi gets from Espressif but if it is not much then the lack of support for the older library version may not matter.

Aside from support, I am not aware of any improvements in the current esp32 library that would benefit our robotics experiences with Bittle.  My impression is that many changes are due to Espressif's need to support new features on their more recent chips, something that Bittle users would not benefit from.  However, if there is a significant defect in the older esp32 library that affects Bittle, I certainly would like to hear about it.  

As for the Arduino IDE nagging to do updates for the IDE, the boards and the libraries, I just turned that preference off as follows

- version 1.8.x:  File > Preferences and then uncheck the "Check for Updates" option.
- version 2.x:  F1 > Preferences: Open Settings (UI) > search for "arduino.checkForUpdates" and then uncheck the "Check for Updates" option
(see https://forum.arduino.cc/t/i-need-to-prevent-arduino-ide-2-x-from-checking-for-library-updates/1190545)

My Arduino configuration is static and has been for at least 2 years.  The only changes I make are to my copy of the OpenCatEsp32 source code and, so far, I haven't needed to update anything in the Arduino IDE during that time.  I am very much an "if it's not  broken, don't fix it" kind of person.

borntoleave, 2024-08-03:
I don't plan to upgrade to the new ESP32 configuration soon. You're right
that the updates are primarily for Espressif's new features, which we are
not using.

Regarding Arduino IDE, it's relatively stable with updates. Both 1.8.x and
2.x work. Most of the changes are in the UI.

FlexoTim, 2024-08-03:
I wasn't requesting to upgrade the library.
I was requesting the detection of the required library.
This checks that the esp32 library version is 2.0.12

#if ESP_ARDUINO_VERSION != ESP_ARDUINO_VERSION_VAL(2, 0, 11)
#error esp32 v 2.0.12 required
#endif

este-este, 2024-08-03:
Got it!  That sounds like a good idea to me, though I assume you mean:
#if ESP_ARDUINO_VERSION != ESP_ARDUINO_VERSION_VAL(2, 0, **_12_**)

FlexoTim, 2024-08-03:
I tested what is in the header esp_arduino_version.h
The patch number for v2.0.12 is 11.

este-este, 2024-08-03:
Well, that is unexpectedly confusing.  Thanks for figuring that out!