PetoiCamp/OpenCatEsp32-Quadruped-Robot #29: std::string rxValue = pCharacteristic->getValue(); compile issue

sub2zero, 2025-03-27:
In file included from ~/OpenCatEsp32/src/OpenCat.h:638,
                 from ~/OpenCatEsp32/OpenCatEsp32.ino:28:
~/OpenCatEsp32/src/bleUart.h: In member function 'virtual void MyCallbacks::onWrite(BLECharacteristic*)':
~/OpenCatEsp32/src/bleUart.h:54:52: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
   54 |     std::string rxValue = pCharacteristic->getValue();
      |                           ~~~~~~~~~~~~~~~~~~~~~~~~~^~
exit status 1
Error compiling for board ESP32 Dev Module.

during compile sketch OpenCatEsp32.ino

borntoleave, 2025-03-27:
Please use ESP32 board version 2.0.12 in Arduino IDE to compile.

sub2zero, 2025-03-27:
thx, but there are another error comes:
`In file included from ~/OpenCatEsp32/src/icm42670/ICM42670P.h:23,
                 from ~/OpenCatEsp32/src/icm42670/ICM42670P.cpp:19:
~/Library/Arduino15/packages/esp32/hardware/esp32/2.0.12/libraries/Wire/src/Wire.h: In function 'int i2c_read(inv_imu_serif*, uint8_t, uint8_t*, uint32_t)':
~/Library/Arduino15/packages/esp32/hardware/esp32/2.0.12/libraries/Wire/src/Wire.h:127:13: note: candidate 1: 'uint8_t TwoWire::requestFrom(int, int)'
     uint8_t requestFrom(int address, int size);
             ^~~~~~~~~~~
~/Library/Arduino15/packages/esp32/hardware/esp32/2.0.12/libraries/Wire/src/Wire.h:125:13: note: candidate 2: 'uint8_t TwoWire::requestFrom(uint8_t, uint8_t)'
     uint8_t requestFrom(uint8_t address, uint8_t size);
             ^~~~~~~~~~~
Sketch uses 1280841 bytes (97%) of program storage space. Maximum is 1310720 bytes.
Global variables use 46332 bytes (14%) of dynamic memory, leaving 281348 bytes for local variables. Maximum is 327680 bytes.`
Does it mean that Partition Scheme should be switched from default 4Mb to something different?

I'm usei Bittle X

este-este, 2025-03-27:
Use the following to give you more program memory:
[image]

Also, it is a bit outdated but you might find this post useful:  https://www.petoi.camp/forum/basic-assembly-and-setup/my-bittle-biboard-software-setup-instructions

closed 2025-03-28 by sub2zero